NubeIO/lora-raw #74: Lora Sensor Names should allow spaces

CanuckMarc, 2021-04-07:
Currently the names of the LORA sensors do not allow names with spaces.  

Can we support names with spaces?

NubeDev, 2021-04-07:
Its done like that for topics on MQTT

maybe we need to add a display name

CanuckMarc, 2021-04-07:
maybe we can automatically convert spaces to underscores?   on import and when the user enters the name manually.

RaiBnod, 2021-04-23:
Our convention for name: alphanumeric values with other characters `-` and `_` on all over the places.

We query from name sometimes, and having space becomes weird query API.

So, I wanna stick with current naming convention (don't wanna do magical backend operation: converting spaces to underscore util we must needed). It's showing proper message though.

closed 2021-04-23 by CanuckMarc