NubeIO/lora-raw #80: Remove scaling and offset input fields for LORA droplet sensors.

CanuckMarc, 2021-04-11:
Lora Droplets don't need `Scaling` or `Offset` input fields.  `Scaling` is not neccesary for any of the droplet sensors, and the `Offset` function will be replaced by the `Math Expression` function.

`Scale` function is still required for `MicroEdge` devices.

NubeDev, 2021-04-11:
Already done just will do a build soon
https://github.com/NubeIO/rubix-plat/pull/561

RaiBnod, 2021-04-23:
Math operation is applied. However, we don't wanna have that tick, if we don't need math operation we just clears it.

Plus, those min, max positions are swapped. And scaling is also needed as per our latest pull request.

Hence, I am closing down this ticket. If we needed it later, we can re-open.