NubeIO/lora-raw #91: MQTT persistance dosnt seem to be working

NubeDev, 2021-04-19:
Also see https://github.com/NubeIO/rubix-wires/issues/1348

CanuckMarc, 2021-04-19:
I'm reasonably certain that the `retain` flag is our issue.  This needs to be set for all messages that need to be retrieved on reboot, or re-subscribe.   

I tested all the other mosquitto configurations, and didn't have any success.

RaiBnod, 2021-04-20:
In my local system, it's persisting the value when we have config `mqtt` > `retain: True`. See the video:

https://user-images.githubusercontent.com/6800775/115340466-7268cf00-a1c6-11eb-9bf8-a562d6210113.mov

NubeDev, 2021-04-20:
@RaiBnod i can show you there is an issue

RaiBnod, 2021-04-23:
@NubeDev and I debugged it and it's working fine.

It's may be related to the wires nodes issue. So, we can close this ticket.

RaiBnod, 2021-05-25:
Old fixed issue, hence closing down this ticket.

closed 2021-05-25 by RaiBnod